acecilia/OpenWRTInvasion #178: MI Router 4A (Gigabytes) Version 2.30.500 is not supported

LinYKen, 2023-06-19:
OpenWRTInvasion % python3 remote_command_execution_vulnerability.py
/Users/yukun/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Router IP address [press enter for using the default 'miwifi.com']: 
Enter router admin password: xxxxxxxxx
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: miwifi.com
stok: xxxxxxxxxxxxxxxxxxxxxxxxx
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:57270. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet miwifi.com

pvtmp, 2023-12-10:
Same issue here ;( did you fix it??

Mike-Wei, 2024-01-17:
You can check out #141. Follow the guide on how to use nc to get a busybox with telnetd support to the system. Then you can open telnetd to access the shell.

Basically, the problem with the stock firmware is that it has dropbear removed. Though the stock busybox has telnetd packaged in, it is difficult to get it to run (at least in my case). So the solution is to get a newer version of busybox with telnetd properly set up already.

A few tips and caveats: 
1. The device with 2.30.x is r4av2. Differentiate it with the 2.28.x version.
2. The isa is mips32r2. The busybox in #141 doesn't work for me. I had to compile my own busybox with several tries and errors.
3. You can always use wget or curl to download things, instead of having to split busybox and using nc.

webysther, 2024-04-08:
Works, please go to: https://github.com/acecilia/OpenWRTInvasion/pull/155#issuecomment-2043692983

Earls1996, 2024-07-19:
> 2. The isa is mips32r2. The busybox in [Not working R4A (Xiaomi 4a Gigabit) 2.30.20 #141](https://github.com/acecilia/OpenWRTInvasion/issues/141) doesn't work for me. I had to compile my own busybox with several tries and errors.
> 3. You can always use wget or curl to download things, instead of having to split busybox and using nc.

Hello, could you please share compiled working busybox ? (with OpenWRT I guess)
2.30.28 firmware R4aG/V2 and it doesn't seem to launch
Thanks

Earls1996, 2024-07-19:
Same issue, r4ag v2 chinese edition, updated to 2.3.500 from 2.30.28 but script won't work on ubuntu (booted on separate pc)  Tried URL execution [https://github.com/acecilia/OpenWRTInvasion/issues/141#issuecomment-1465561959](url) , but busybox presented in post won't seem to start after executing 'chmod a+x /tmp/split/$$/tmp/split telnetd`  cause telnet won't accept connection.

DmitryBLKV, 2024-08-18:
**Issue: SSH connection not working after script execution**

Hi,

I’m experiencing issues when trying to run the OpenWRT Invasion script on my Xiaomi Mi Router 4A Gigabit Edition (Firmware version 2.30.500).

After executing the script, I receive the following message:

`dmitrybelyakov@dmitry-5 OpenWRTInvasion % python3 remote_command_execution_vulnerability.py
/Users/dmitrybelyakov/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:35: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Router IP address [press enter for using the default 'miwifi.com']: 
Enter router admin password: 
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: miwifi.com
stok: 58c52d49be63014309e89ba7c3104bb4
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:61529. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet miwifi.com`

I've tried using Telnet, but I'm getting "Connection refused". 

Does anyone have any advice on how to resolve this, or is there something specific I should check with my router's firmware version?

Thanks for your help!

[image]