acecilia/OpenWRTInvasion #181: Unpacking failed. The file may be damaged.

EvgeniyBlinov, 2023-10-13:
MIWIFI 3C(R3L) Chinese
2.14.45
OpenWRTInvasion fcec03a49d78d700d62f7be82093bd8e349d9a55

```
r1 = requests.post(
    "http://{}/cgi-bin/luci/;stok={}/api/misystem/c_upload".format(router_ip_address, stok),
    files={"image": open("build/payload.tar.gz", 'rb')},
    proxies=proxies
)
print(r1.text)
```

Translated
```
start uploading config file...
{"code":1629,"msg":"Unpacking failed. The file may be damaged"}
```

```
file ./payload.tar.gz 
./payload.tar.gz: gzip compressed data, was "payload.tar", last modified: Fri Oct 13 10:04:39 2023, max compression, original size modulo 2^32 10240
```

acecilia, 2023-10-15:
Yep, network configuration error. Try modifying your network configuration or read similar cases in this repo

closed 2023-10-15 by acecilia

EvgeniyBlinov, 2024-02-01:
@acecilia issue closed, but I don't see any changes.
`{"code":1629,"msg":"Unpacking failed. The file may be damaged"}` - not marked as expected error.