acecilia/OpenWRTInvasion #186: Warning: the process has finished, but seems like ssh connection to the router is not working as expected.

Benjeny, 2024-01-28:
Hello, is there someone who can help me ?
First of all, thank you for your work.
I am trying to use the exploit on a Xiaomi Mi4A (Gigabit Edition) router, chinese version, firmware Version 2.28.62
The goal is to flash openwrt.

I have connected my MAC directly to the Mi router (192.168.31.1),followed the video(https://www.youtube.com/watch?v=SLbkce-M2nE&list=PL_Z5kQQ5KSSXOenNFFUOq2qMdOYEzRngq&index=12) of Youtuber, Hoddys Guides,steped by steped. When the viedo at  12:22, showed that"Warning: the process has finished, but seems like ssh connection to the router is not working as expected....", what happened? how and why?

freddysolorzano, 2024-01-28:
I have the same problem but with an r4a router manufactured in 2023.5 with firmware 2.30.28

AddaxSoft, 2024-04-05:
same on 2.30.28; this exploit command seems to be working and it does reboot the router, but when running the full exploit chain it fails.

`http://192.168.31.1/cgi-bin/luci/;stok={{{STOK}}}/api/misystem/set_config_iotdev?bssid=XXXXXX&user_id=XXXXXX&ssid=-h%0Areboot%0A`

RadioOperator, 2024-04-05:
If fw 2.30.28 is the same hardware (Xiaomi 4A v2) with 2.30.20, maybe this issue would help:
https://github.com/acecilia/OpenWRTInvasion/issues/141

AddaxSoft, 2024-04-05:
yes it does help as I already mentioned above the initial RCE exploit works (the router reboots) but when the exploit is chained (binary upload, and trying to execute it later for telnet access) something breaks in between

RadioOperator, 2024-04-05:
try to downgrade to 2.30.20 firmware.

licryle, 2024-09-03:
I just went through the steps in the mentioned thread but found them too complicated. So created a much easier shell script for 2.30.28 and documented the commands to send, check out https://github.com/acecilia/OpenWRTInvasion/issues/141#issuecomment-2326266325.

I'm now looking for the right firmware to not brick the device :)