acecilia/OpenWRTInvasion #191: Xiaomi Router 4A Gigabit Edition fails to SSH

tinatsina, 2024-06-02:
I have been trying to upload OpenWRT Invasion for some time, but it seems like I end up with the same error regardless of what I do.

`warning: the process has finished. But it seems like the ssh connection to the router is not working as expected`

I have looked into the issues, and none of them have helped so far.

1. Changed the DHCP range to 2-5
2. I Tried a proxy server, but it also did not work.

Using MiWiFi firmware version 2.3.500

sensiblepuffin, 2024-06-07:
I didn't try ssh'ing, but I had success telnet'ing in and downloading the binary directly into /tmp.

FeOAr, 2024-06-14:
My system version is 2.28.69, and I have the same issue.

FeOAr, 2024-06-14:
#185  Through this method, my device is now working.