acecilia/OpenWRTInvasion #194: ssh: connect to host 192.168.31.1 port 22: Connection refused

max-end, 2024-08-17:
r4a version: 2.30.28

Router IP address [press enter for using the default 'miwifi.com']:
Enter router admin password: 12345678
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: miwifi.com
stok: c95a79437de2ca690ee3e53797533537
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:39891. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet miwifi.com
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@miwifi.com
* ftp: using a program like cyberduck

ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@miwifi.com
ssh: connect to host miwifi.com port 22: Connection refused

telnet 192.168.31.1
Connecting To 192.168.31.1...Could not open connection to the host, on port 23: Connect failed

ssh root@192.168.31.1
ssh: connect to host 192.168.31.1 port 22: Connection refused

XZC1516, 2024-10-26:
I have the same issue, mine is running with docker and I turn off the firewall but link 22 is also failing

MonsterSe7en, 2024-11-08:
Same issue.... How do I resolve this
---

C:\Users\Monster>docker run --network host -it openwrtinvasion
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: rootroot
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: 192.168.31.1
stok: f01f929538235058c1e9ff4f909c9691
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:53761. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1

---

C:\Users\Monster>telnet 192.168.31.1
Connecting To 192.168.31.1...Could not open connection to the host, on port 23: Connect failed

taxtigerno1, 2025-02-16:
> r4a version: 2.30.28
> 
> Router IP address [press enter for using the default 'miwifi.com']: Enter router admin password: 12345678 There two options to provide the files needed for invasion:
> 
> 1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
> 2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
>    Which option do you prefer? (default: 1)
> 
> router_ip_address: miwifi.com stok: c95a79437de2ca690ee3e53797533537 file provider: local file server
> 
> start uploading config file... start exec command... local file server is runing on 0.0.0.0:39891. root='script_tools' done! Now you can connect to the router using several options: (user: root, password: root)
> 
> * telnet miwifi.com
> * ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null [root@miwifi.com](mailto:root@miwifi.com)
> * ftp: using a program like cyberduck
> 
> ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null [root@miwifi.com](mailto:root@miwifi.com) ssh: connect to host miwifi.com port 22: Connection refused
> 
> telnet 192.168.31.1 Connecting To 192.168.31.1...Could not open connection to the host, on port 23: Connect failed
> 
> ssh root@192.168.31.1 ssh: connect to host 192.168.31.1 port 22: Connection refused

我也碰到相应问题，显示可以通过三种方式连接，但是我使用ssh连接显示拒绝，连接不了。

taxtigerno1, 2025-02-16:
> r4a version: 2.30.28
>
> Router IP address [press enter for using the default 'miwifi.com']: Enter router admin password: 12345678 There two options to provide the files needed for invasion:
>
> 1. Use a [...]

can you solved this problem?