acecilia/OpenWRTInvasion #197: [Failed] New vesion of Xiaomi 4A

Pull256, 2024-09-25:
I just bought a Xiaomi 4A Gigabit in JingDong(JD). It seems like Xiaomi has changed the version of 4A.
The SKU number of this router is DVB4397CN and the firmware is 1.0.X instead of  2.X or 3.X.
I failed to hack this Router and if anybody want to buy a brand new 4A, Please watch out the version.

doom345, 2024-10-01:
hello, I encountered with the same trouble. Have u figured it out. thank u

medvedicos, 2024-11-15:
Hi. I got one of these devices in my hands too. No luck breaking in.

Pull256, 2024-12-07:
Eventually, I got ssh access through this amazing program [xmir-patcher](https://github.com/openwrt-xiaomi/xmir-patcher).
However I found that the CPU of this router is not mt7621 but RTL8197F. Below is the message from  _cat /proc/cpuinfo_
```
root@XiaoQiang:~# cat /proc/cpuinfo
system type             : RTL8197F
machine                 : Unknown
processor               : 0
cpu model               : MIPS 24Kc V8.5
BogoMIPS                : 666.41
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 64
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r2
ASEs implemented        : mips16
shadow register sets    : 4
kscratch registers      : 0
package                 : 0
core                    : 1
VCED exceptions         : not available
VCEI exceptions         : not available
```

> [!WARNING]
> So I'm guessing that this CPU won't work with existing breeds made for mt7621, and since I won't be able to use TTL to debrick, I'm giving up on continuing to flash it in. 建议不要刷入breed，CPU型号不符，怀疑小米改了制作工艺！

JimAetos, 2025-02-22:
I got the XIAOMI R4A Gigabit SKU: DVB4397CN also, manufacture date 2024/10
In the web interface says Firmware 1.0.34
No luck with /acecilia/OpenWRTInvasion.
I opened the case , the board marks MTO-WRMD05 ,the CMOS marks DM AY2424 25Q128SIG ENG836,  the central chip is RTL8367RB (i suppose it servers the switching) ,the 2.4Ghz  side has a RTL8197FH and the  5Ghz a RTL8812FR.
The board is shielded by a big metal plate , very solid construction.
I wish someone finds a way to openwrt it... or at least a way to load a global firmware so the interface turns to Greek or English.

wb1016, 2025-04-11:
i was also baited to buy this without knowledge. i tried OpenWRTInvasion but it did not work, and tried switching language at least, and also tried reverse-engineering the firmware image. i could not find any firmware image related to this specific version, nor found way to switching language. and most importantly, this has the famous RTL8197F SoC: the cpu with realtek version of MIPS architecture. because they are too different than mainline MIPS, it will never get OpenWRT support. and furthermore, the RAM was downgraded than previous version: which conflicts with 8/64 limit.
i will not dig into this.