acecilia/OpenWRTInvasion #198: [FAILURE] Mi Router 4A 100M on firmware 3.0.129 (R4AC)

justbendev, 2024-10-02:
#### Hi everyone ! :wave: 

Tried to get a shell with v0.0.1 first since i didn't want to connect the router to internet but it failed.
I then tried the latest (master) [fcec03a](https://github.com/acecilia/OpenWRTInvasion/commit/fcec03a49d78d700d62f7be82093bd8e349d9a55) but it also failed.

Tried to downgrade to a known compatible version but it won't let you downgrade _"for security reasons"_
Due to environment constrains i can't use any Windows machine so i can't use a "Debricking tool to force downgrade" since they are only compatible with Windows / Mac

No TFTP documentation anywhere online for this specific modem.

```
VM@linux:~/OpenWRTInvasion$ python3 remote_command_execution_vulnerability.py 
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: REDACTED
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: 192.168.31.1
stok: ee3b2902bbeb22e7b0a5916a093c1924
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:43135. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1
```

justbendev, 2024-10-03:
Apparently i have the V2 version of this router.
Got a dump of alot of useful info by doing a device backup on the Xiaomi Web UI.

```
Filesystem                Size      Used Available Use% Mounted on
rootfs                   11.0M     11.0M         0 100% /
/dev/root                11.0M     11.0M         0 100% /
tmpfs                    29.6M     11.3M     18.3M  38% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mtdblock9            2.2M    208.0K      2.0M   9% /userdisk
/dev/mtdblock6            1.0M    400.0K    624.0K  39% /data
/dev/mtdblock6            1.0M    400.0K    624.0K  39% /etc
/dev/root                 1.0M    400.0K    624.0K  39% /mnt
/dev/mtdblock6            1.0M    400.0K    624.0K  39% /mnt
==========bootinfo

ROM    ver: config core 'version'
	# ROM ver
	option ROM '3.0.129'
	# channel
	option CHANNEL 'release'
	# hardware platform R1AC or R1N etc.
	option HARDWARE 'R4ACv2'
	# CFE ver
	option UBOOT '1.0.0'
	# Linux Kernel ver
	option LINUX '0.0.1'
	# RAMFS ver
	option RAMFS '0.0.1'
	# SQUASHFS ver
	option SQAFS '0.0.1'
	# ROOTFS ver
	option ROOTFS '0.0.1'
	#build time
	option BUILDTIME 'Wed, 14 Sep 2022 13:18:00 +0000'
	#build timestamp
	option BUILDTS '1663161480'
	#build git tag
	option GTAG 'commit 4062d54ed1be05d43a2e1d2bca550a29cbff355b'
Hardware  : Ver. A
ROM    sum: 
System    : Dual - 1
KERNEL    : console=ttyS1,115200n8 uart_en=0 factory_mode=0 mem=64m root=/dev/mtdblock8

MTD  table:
dev:    size   erasesize  name
mtd0: 01000000 00010000 "ALL"
mtd1: 00020000 00010000 "Bootloader"
mtd2: 00010000 00010000 "NULL"
mtd3: 00010000 00010000 "Factory"
mtd4: 00010000 00010000 "crash"
mtd5: 00010000 00010000 "cfg_bak"
mtd6: 00100000 00010000 "overlay"
mtd7: 00c60000 00010000 "OS1"
mtd8: 00b00000 00010000 "rootfs"
mtd9: 00230000 00010000 "disk"
mtd10: 00010000 00010000 "Config"
```

sudoatp, 2024-10-30:
I have the same router and the same problem, did you find a way to solve this?

justbendev, 2024-10-30:
@sudoatp I ended up getting a shell **BUT** flashing the OpenWRT Firmware for RA4Cv2 **bricked the device.**

And since i didn't make a backup of original firmware i couldn't use it with XiaomiRepairTool on a VM.
Xiaomi Firmware are older than original firmware and flahsing thoses didn't unbrick the device even with a sucessfull blue led blinking indicating a successful reflash

Either way, first you can try setting your router as a WiFi Repeater connected to WiFi with Internet access because in Router mode it will fail.
Then try the master branch again BUT with OPTION 2 to download the payload from the internet instead of from the local server.

If that fail you can try again with the pull request branch but not all RA4Cv2 will be compatible with that one. 

Im doing all of this 6000Km away from the physical hardware so i will get back to this thread this weekend when i make a copy from a Factory device firmware 3.0.129 International (Global) will publish here a link for thoses who also have a bricked device. And will make step by step instructions.

This device share alot of things with other Xiaomi devices so i won't be hard to figure this out.

jefcolbi, 2024-11-12:
Hi @justbendev
any update about this issue?

Neustradamus, 2025-04-21:
Any news about the file search?

justbendev, 2025-04-21:
@Neustradamus  , I used the official Xiaomi recovery tool it internally uses TFTP and it fetches the firmware automatically.

Obtaining a shell was possible you just had to setup the router in the correct mode; otherwise the exploit failed.

At the time i used a special snapshot built with the correct partition offsets, for a different router (4C i think) and then upgraded to 4A. **Now this is unnecessary because a proper firmware made for this specific device is available.**