acecilia/OpenWRTInvasion #202: Cannot Unlock Telnet on MiRouter R4A with ROM 2.28.62

Coconut-Aero, 2024-11-14:
It's output seemed like:
```
(venv) python3 remote_command_execution_vulnerability.py 
Router IP address [press enter for using the default 'miwifi.com']: 
Enter router admin password: ###MYPASSWD###
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: miwifi.com
stok: ###MYSTOK###
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:43881. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet miwifi.com               
```
When I bought this router its 2.28.65 and I didn't connect it to the Internet.
I configured manually and flashed 2.28.62.
I can't telnet it or ssh it.
It seems to have a wrong setting of time but I fixed it later.

xuzuxing, 2025-03-06:
The new Xiaomi 4A Gigabit v2

```
wget https://github.com/acecilia/OpenWRTInvasion/archive/0cc4fce4e38eb23c761d516942fa67762c4d4b26.zip
```

python3 remote_command_execution_vulnerability_v2.py

Coconut-Aero, 2025-03-07:
Thank you for your reply. I have resolved this issue by downgrading my ROM, referring to #200.

closed 2025-03-07 by Coconut-Aero