acecilia/OpenWRTInvasion #206: Mi Router AC1200 EU (rb02) on firmware 3.0.61

Amirali-Foroutan, 2025-03-20:
I’ve followed all the instructions and videos, and even read some forums, but I still couldn’t succeed with the 3.0.61 firmware on my Mi Router AC1200 EU (RB02) it just refuse to start telnet or ssh the ports 22 and 23 are also closed. However, it seems some people have succeeded with the exact same model [#192 ](https://github.com/acecilia/OpenWRTInvasion/issues/192) and [#155 ](https://github.com/acecilia/OpenWRTInvasion/pull/155#issuecomment-2143710960) someone help me here on how to do it properly?

Amirali-Foroutan, 2025-03-21:
Solution: Installing OpenWRT 23.05.2 on Mi Router AC1200 EU (rb02) - Firmware 3.0.61
Note: This device is identical to the Mi Router 4A Gigabit V2. The process involves exploiting a vulnerability using a Windows machine and then accessing the router via Telnet from an Ubuntu VM (e.g., on VMware).

Tools and Resources:
Exploit Source: https://github.com/LordPinhead/OpenWRTInvasion

OpenWRT Firmware: http://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/openwrt-23.05.2-ramips-mt7621-xiaomi_mi-router-4a-gigabit-v2-squashfs-sysupgrade.bin

Operating Systems Used:
Windows (to run the exploit)
Ubuntu (via VMware, for Telnet access)

Preparation:
Modify the Exploit Script for Windows Compatibility:
Open remote_command_execution_vulnerability.py from the OpenWRTInvasion repository.

Remove the line: sys.exit("Stopping: script can only be run on a Mac/Linux system").

Save the file. This allows the script to run on Windows.

Steps:
Part 1: Run the Exploit on Windows
Open a terminal (e.g., Command Prompt or PowerShell) on your Windows machine.
Navigate to the directory containing remote_command_execution_vulnerability.py.

Execute the following command:
python remote_command_execution_vulnerability.py
This exploits the vulnerability and enables Telnet access on the router.

Part 2: Access the Router via Telnet on Ubuntu
Launch your Ubuntu virtual machine (e.g., via VMware).
Open a terminal in Ubuntu.
Connect to the router using Telnet with the following command:
telnet 192.168.31.1

Log in with the default credentials:
Username: root
Password: root

Part 3: Install OpenWRT Firmware
Once logged in, navigate to the /tmp directory:
cd /tmp

Download the OpenWRT firmware file:
wget http://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/openwrt-23.05.2-ramips-mt7621-xiaomi_mi-router-4a-gigabit-v2-squashfs-sysupgrade.bin -O firmware.bin

Verify the integrity of the downloaded file (optional but recommended):
./busybox sha256sum firmware.bin

Flash the firmware to the router:
mtd -e OS1 -r write firmware.bin OS1
This erases the existing OS1 partition, writes the new firmware, and reboots the router.

closed 2025-03-21 by Amirali-Foroutan

Neustradamus, 2025-04-21:
@acecilia: Have you seen?

SteffenR87, 2025-11-11:
I have been using Windows 11 physically on machine, no VM. Antivirus and Firewall turned off. I tried docker and also python on win11 directly. Script worked, logged into router but wasn't able to open telnet.

SteffenR87, 2025-11-12:
Today I tried with Linux. Same issue. I am not able to upload file. Any ideas? I bought several devices with intention of upgrading to openwrt